paviro/MMM-Facial-Recognition #5: Editing config.js

vanimesh76, 2016-04-26:
When I edit the config.js the web page returns a bank screen 
I am running it using "node serveronly"

paviro, 2016-04-26:
Check your web inspector you probably formatted something wrong.

vanimesh76, 2016-04-26:
I don't get it, what should I look for in the web Inspector to correct, 

Is this the only part I need to add in the "config.js" of MagicMirror

{
    module: 'MMM-Facial-Recognition',
    config: {
        // 1=LBPH | 2=Fisher | 3=Eigen
        RECOGNITION_ALGORITHM: 1,
        // Threshold for the confidence of a recognized face before it's considered a
        // positive match.  Confidence values below this threshold will be considered
        // a positive match because the lower the confidence value, or distance, the
        // more confident the algorithm is that the face was correctly detected.
        LBPH_THRESHOLD: 50,
        FISHER_THRESHOLD: 250,
        EIGEN_THRESHOLD: 3000,
        // Path to your training xml
        TRAINING_FILE: 'modules/MMM-Facial-Recognition/training.xml',
        // recognition intervall in seconds (smaller number = faster but CPU intens!)
        INTERVAL: 2,
        // Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
        LOGOUT_DELAY: 15,
        // Array with usernames (copy and paste from training script)
        USER: [],
        //Module set used for strangers and if no user is detected
        DEFAULT_CLASS: "default",
        //Set of modules which should be shown for every user
        FOR_ALL: "for_all"
    }
}

vanimesh76, 2016-04-26:
when i paste this screen goes blank and black, but removing it brings it back to normal

paviro, 2016-04-26:
That's because you did a mistake when adding it. You have to add it to the module array. In the web inspector you will see an error and what the problem is.

paviro, 2016-04-26:
Don't forget, when there is any module after the facial recognition module, you have to add a comma at the end!

vanimesh76, 2016-04-26:
Am I supposed to add this in the defaultmodule.js
var defaultModules = [
	"alert",
	"calendar",
	"clock",
	"compliments",
	"currentweather",
	"helloworld",
	"newsfeed",
	"weatherforecast"
        "MMM-Facial-Recognition"
];

vanimesh76, 2016-04-26:
I paseted that part with a comma and also updated the defaultmodule.js
But it just shows a blank screen, is it supposed to detect my face and then turn the screen on?

vanimesh76, 2016-04-26:
And what is a web inspecter where i need to paste it, is it like the "Inspect Elment" by Chrome

paviro, 2016-04-26:
No!

paviro, 2016-04-26:
No it is not. Again! Check your web inspector the error will be displayed there!

paviro, 2016-04-26:
Press alt (if on Linux and probably Windows), it will show the menus. In one of them you'll find the development tools, open them and you get a web inspector.

vanimesh76, 2016-04-26:
ok

vanimesh76, 2016-04-26:
Ok after cloning Te Facial recognition code in the modules folder.
where am I supposed to run the "npm Install" command in the same folder or the root folder MagicMirror

paviro, 2016-04-26:
In the cloned folder "MMM-Facial-Recognition".

vanimesh76, 2016-04-26:
ok I did that part in the root folder

vanimesh76, 2016-04-26:
ok The facial recognition system seems to be working as it shows hello stranger alert. :)

but The screen is still a Blank black page, nothing to be found in the Web Inspector

the command promt server shows that all other modules have executed

Create new calendar fetcher for url: http://www.calendarlabs.com/templates/ical/US-Holidays.ics - Interval: 300000
Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
[MMM-Facial-Recognition] Facerecognition started...
[MMM-Facial-Recognition] Loading training data...
[MMM-Facial-Recognition] ALGORITHM: LBPH
[MMM-Facial-Recognition] Training data loaded!
[MMM-Facial-Recognition] --------------------
[MMM-Facial-Recognition] Beutztes System: AMD64
[MMM-Facial-Recognition] Webcam ausgewählt...
[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.
[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.
[MMM-Facial-Recognition] User undefined with confidence null logged in.
Use exsisting news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml
[MMM-Facial-Recognition] User undefined logged out.

paviro, 2016-04-26:
Can you share you config.js?

vanimesh76, 2016-04-26:
ok sharing...and great news Face recognition is now working perfectly :)

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'en',
	timeFormat: 24,
	units: 'metric',

	modules: [
		{
			module: 'alert',
		},
		
		{
			module: 'clock',
			position: 'top_left'
		},
		{
			module: 'calendar',
			header: 'US Holidays',
			position: 'top_left',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics'
					}
				]
			}
		},
		{
			module: 'compliments',
			position: 'lower_third'
		},
		{
			module: 'currentweather',
			position: 'top_right',
			config: {
				location: 'New York',
				appid: 'YOUR_OPENWEATHER_API_KEY'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			header: 'Weather Forecast',
			config: {
	            location: 'New York',
	            appid: 'YOUR_OPENWEATHER_API_KEY'
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_bar',
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
		{
    module: 'MMM-Facial-Recognition',
    config: {
        // 1=LBPH | 2=Fisher | 3=Eigen
        RECOGNITION_ALGORITHM: 1,
        // Threshold for the confidence of a recognized face before it's considered a
        // positive match.  Confidence values below this threshold will be considered
        // a positive match because the lower the confidence value, or distance, the
        // more confident the algorithm is that the face was correctly detected.
        LBPH_THRESHOLD: 50,
        FISHER_THRESHOLD: 250,
        EIGEN_THRESHOLD: 3000,
        // Path to your training xml
        TRAINING_FILE: 'modules/MMM-Facial-Recognition/training.xml',
        // recognition intervall in seconds (smaller number = faster but CPU intens!)
        INTERVAL: 2,
        // Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
        LOGOUT_DELAY: 15,
        // Array with usernames (copy and paste from training script)
        USER: ['Ani'],
        //Module set used for strangers and if no user is detected
        DEFAULT_CLASS: "default",
        //Set of modules which should be shown for every user
        FOR_ALL: "for_all"
    }
},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

vanimesh76, 2016-04-26:
**Voila!!!** Found the catch.
In "MMM-Facial-Recognition.js" 

**Comment out this section**
`// notificationReceived: function(notification, payload, sender) {
	// 	if (notification === 'DOM_OBJECTS_CREATED') {
	// 		MM.getModules().exceptWithClass("default").enumerate(function(module) {
	// 			console.log(module);
	// 			module.hide(1000, function() {
	// 				Log.log('Module is hidden.');
	// 			});
	// 		}
	// 	}
	// },`

:)

paviro, 2016-04-26:
Oh I see... You did not read the README carefully! You have to add custom classes for the module to be useful including "for_all" and "default" else anything gets hidden. Do not comment out this section or else the module will be useless...

vanimesh76, 2016-04-26:
oh..ok
so I need to add "for_all" and "default" in config.js like the first part?

paviro, 2016-04-26:
It is all explained [here](https://github.com/paviro/MMM-Facial-Recognition#usage).

closed 2016-04-30 by paviro