paviro/MMM-Facial-Recognition #6: facerecognition

vanimesh76, 2016-04-26:
Face recognition not working on the raspberry pi3.
neither the capture.py

ERROR

Iam using The latest version.

Images will be placed in ./training_data/sde

--------------------
Starting process...

Traceback (most recent call last):
  File "capture.py", line 35, in <module>
    capture.capture()
  File "/home/pi/Downloads/MMM-Facial-Recognition-Tools-master/facetrainer/lib/capture.py", line 37, in capture
    camera = config.get_camera()
  File "/home/pi/Downloads/MMM-Facial-Recognition-Tools-master/facetrainer/lib/config.py", line 39, in get_camera
    return webcam.OpenCVCapture(device_id=0)
  File "/home/pi/Downloads/MMM-Facial-Recognition-Tools-master/facetrainer/lib/webcam.py", line 29, in __init__
    self._camera.open()
TypeError: Required argument 'device' (pos 1) not found

paviro, 2016-04-27:
In the config.py see the get_camera function. It checks what device you are using it supposes your are using a computer.. As you can see it will print a string with the name of the system architecture. You have to add it in ""arm" in _platform:" instead of arm. Please tell me what it is so I can add it.

paviro, 2016-04-30:
Any progress here?

closed 2016-04-30 by paviro

vanimesh76, 2016-12-03:
Thanks paviro for the help, everything worked very well.
Sorry I could'nt inform you in time.
The Project worked just fine.
:)