paviro/MMM-Facial-Recognition #7: Camera is not enabled

PtrBld, 2016-04-30:
Hi 
I am using a raspberry Pi 3 (finally) and am figuring the following error when starting my MM:
`mm-0 (err): { [Error: mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mm-0 (err): mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mm-0 (err): Traceback (most recent call last):
mm-0 (err):   File "modules/MMM-Facial-Recognition/facerecognition/facerecognition.py", line 72, in <module>
mm-0 (err):     image = camera.read()
mm-0 (err):   File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/facerecognition/picam.py", line 23, in read
mm-0 (err):     with picamera.PiCamera() as camera:
mm-0 (err):   File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 488, in __init__
mm-0 (err):     self.STEREO_MODES[stereo_mode], stereo_decimate)
mm-0 (err):   File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 526, in _init_camera
mm-0 (err):     "Camera is not enabled. Try running 'sudo raspi-config' "
mm-0 (err): picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
`

I did not want to use the camera module at all, but have a USB camera connected to my pi. The facerecognition tester is working great with the connected web cam. Do I need to change some settings in order to make it work with a webcam?
Thanks in advance

paviro, 2016-04-30:
I don't think you can use a USB webcam with the Pi. At least this script can't.

paviro, 2016-04-30:
You can try adding support (some info [here](https://www.raspberrypi.org/documentation/usage/webcams/)), there is a webcam.py you can use as reference.

closed 2016-04-30 by paviro

PtrBld, 2016-04-30:
ok its working with your other script (facerecognition tools) so I will take a look into it.
Thank you.

paviro, 2016-04-30:
Tell me what you changed so I can add it as an option in a future release.

paviro, 2016-04-30:
Check my update I added an option. If usb webcams work on the Pi it should be supported now. See README for configuration key.

PtrBld, 2016-04-30:
It is working great!
Thank you!

paviro, 2016-04-30:
Glad to hear :)