paviro/MMM-Facial-Recognition #9: Separate No User and Unknown User

skoupayou, 2016-05-22:
Hi there,
Again I didn't goes throught the code but I wonder if it's simple to implement the possibility to separate Unknown and No User.
To be able to run case like : 
- No user : just show clock
- Unknown user : show a bit more option 

Regards,

paviro, 2016-05-22:
Yes should be easy I will add it when I have some spare time.

paviro, 2016-07-27:
Just a question: Why would you want to show something different when no one sees it anyway?

skoupayou, 2016-08-08:
Hey there, sorry for the late answer.
So it's simple : lot of features by doing that (and more if we can use kid of user management more than hide / show by user)

Few exemple : 
1) I have my mirror in my entry
  - no user = black screen or just clock or something
  - unknown = words like "hello you" more option like but not full one
  - me / known user : dislay my calendar that I don't want everyone to view it ;)

paviro, 2016-08-08:
But no user = no-one will see it anyway so why have a special option?

closed 2016-11-15 by paviro

toniweser, 2016-11-30:
I would also love this feature. Is it already implemented?

paviro, 2016-11-30:
No because there still wasn't anyone who could tell me why it would be useful 😅

mwkrieger, 2017-05-18:
I'd like to request this as well. 

The 'noone' feature would give me the option to basically 'turn off' the display when no one is home/looking at the MM. This would save the display. 

The 'someone' feature would tell me that a person is looking at the mirror, so then we can display the clock/weather/etc. 

The 'everyone' would show information for all known users. 

Then the named users would show specific information for that specific user. 

This would be killer!