paviro/MMM-Facial-Recognition #10: Face Recognition takes time

bikerboymohit, 2016-06-10:
For some reason after the first couples of runs the module takes considerably a lot of time trying to detect the face. How should this be enhanced.

[MMM-Facial-Recognition] Facerecognition started...
[MMM-Facial-Recognition] Loading training data...
[MMM-Facial-Recognition] ALGORITHM: LBPH
[MMM-Facial-Recognition] Training data loaded!
[MMM-Facial-Recognition] --------------------
[MMM-Facial-Recognition] PiCam ausgewählt...  // from here the module waits here for a long time
[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.
[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.

bikerboymohit, 2016-06-10:
This is my config.js
{
    module: 'MMM-Facial-Recognition',
    position: 'top_left',
    config: {
        recognitionAlgorithm: 1,
        lbphThreshold: 50,
        fisherThreshold: 250,
        eigenThreshold: 3000,
        useUSBCam: false,
        trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
        interval: 2,
        logoutDelay: 15,
        users: ['bikerboy'],
        defaultClass: "default",
        forAllClass: "for_all"
    }
},

paviro, 2016-06-10:
You only need to add to the config what you would like to change. The module first had to load the XML file into memory which takes some time. Also the Pi is not the fastest. For a face to be valid it has to be recognized 3 times to minimize mistakes. So a 2 seconds interval makes 6 seconds in total.

bikerboymohit, 2016-06-10:
But this seems to be taking considerably more than 6 seconds atleast for each read . Between each of this step it takes approx 30sec to a minute.

[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.

paviro, 2016-07-27:
Make sure to check your training file with the tester!

CatoAntonsen, 2016-07-31:
I'm having problem getting the trainer (capture.py) to recognize my face in 95% of my trials. Keeps saying "Could not detect single face". Occationally it recognizes my face. I have good lighting and a clean background.

A few times I get an error message: "Corrupt JPEG data: XXX extraneous bytes before marker 0xd9".

Any ideas what is happening?

paviro, 2016-07-31:
Nop no idea. What camera are you using? When I get that error it is mostly due to lighting.

CatoAntonsen, 2016-07-31:
The camera is the [Raspberry Pi Rev 2.1 Official 8.0 Megapixels HD Camera Module - Green](http://eud.dx.com/product/raspberry-pi-rev-2-1-official-8-0-megapixels-hd-camera-module-green-844433744).

I forgot to mention that I can't find the "capture.pgm" that is referred to in the error message.

paviro, 2016-11-15:
The `capture.pgm` is not created. Haven't removed that message yet.. :D 
Have you tried the tester? Does it recognize a face?

paviro, 2016-12-04:
I close this for now feel free to reopen.

closed 2016-12-04 by paviro