paviro/MMM-Facial-Recognition #12: Umm... Where do things go?

pauliolio, 2016-07-09:
Hi

Many thanks for making this available - Really looking forward to getting this going.
I'm making a real rookie mistake here I'm sure - This is the first module I've tried to add in after getting the modules in the sample config file configured & working.

I've put the main(?) module folder called MMM-Facial-Recognition-master in /home/pi/MagicMirror/modules/default/
I've installed the dependencies OK.
I've used the tool to create the training images.
I've added my 'name' to the facerecognition_tester/lib/config.py file
I've used the trainer to create the XML file.

The next step in the test tool instructions have me stuck though:
"2. Copy your `training.xml` from `tools/facetrainer` into the root folder of this tool."
And having added the module info to the MagicMirror conf file all I get is a blank screen now.

So, some questions please:
Is the MMM-Facial-Recognition-master folder in the right place?
There is no 'tools' folder. Does it have to be in a folder called 'tools'? If so, where should it be?
Where does 'into the root folder of this tool' refer to? I'm assuming the 'facerecognition_tester' folder along with 'facerecognition.py'. Is this correct?
Given where I've got the main app folder, is this line in the MagicMirror conf file correct?  
"trainingFile: 'modules/default/MMM-Facial-Recognition-master/training.xml'"

Separately, but with the same 'have I got stuff in the right folders' problem, when I try to use the test tool I get this error:
pi@MirrorPi3:~/MagicMirror/Tools/facerecognition_tester $ python facerecognition.py
Loading training data...
ALGORITHM: LBPH
Training data loaded!
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 42, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

A write-up of what should be where in relation to the magic mirror root folder would be really, really helpful.

Many thanks,
Pauliolio

paviro, 2016-07-09:
Hi there,
I am sorry but I do not provide support on GitHub this is only for bugs etc. If you need help check out the forum http://forum.magicmirror.builders :)

pauliolio, 2016-07-10:
Hi

Ah - Sorry. Shall do.

Many thanks

closed 2016-07-27 by paviro