paviro/MMM-Facial-Recognition #17: MM failed to start

xdker, 2016-09-10:
Hello, I used the configuration of the sample but there was a problem.

```mm-0 [MMM-Facial-Recognition] Facerecognition started...
mm-0 [MMM-Facial-Recognition] Loading training data...
mm-0 [MMM-Facial-Recognition] ALGORITHM: LBPH
mm-0 Create new todo fetcher for list: inbox - Interval: 60000
mm-0 [MMM-Facial-Recognition] Training data loaded!
mm-0 [MMM-Facial-Recognition] --------------------
mm-0 [MMM-Facial-Recognition] PiCam ausgewählt...
mm-0 Whoops! There was an uncaught exception...
mm-0 { Error: connect ETIMEDOUT 216.58.203.46:80
mm-0     at Object.exports._errnoException (util.js:1008:11)
mm-0     at exports._exceptionWithHostPort (util.js:1031:20)
mm-0     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1080:14)
mm-0   code: 'ETIMEDOUT',
mm-0   errno: 'ETIMEDOUT',
mm-0   syscall: 'connect',
mm-0   address: '216.58.203.46',
mm-0   port: 80 }
mm-0 MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
mm-0 If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
mm-0 Whoops! There was an uncaught exception...
mm-0 { Error: socket hang up
mm-0     at createHangUpError (_http_client.js:252:15)
mm-0     at Socket.socketCloseListener (_http_client.js:284:23)
mm-0     at emitOne (events.js:101:20)
mm-0     at Socket.emit (events.js:188:7)
mm-0     at TCP._handle.close [as _onclose] (net.js:492:12) code: 'ECONNRESET' }
mm-0 MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
mm-0 If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues```

paviro, 2016-09-10:
Don’t think this has anything to do with my module. Something tried to  216.58.203.46:80 which seams to be a Google IP. Please check your internet connection!

xdker, 2016-09-11:
However, I checked the network, can connect to the google. Restart MM or no start and no error screen, black screen. Remove this module and it can start

paviro, 2016-09-11:
Can you send me your configuration file?

xdker, 2016-09-12:
<pre><code>		//人脸识别模块
		/*{
    module: 'MMM-Facial-Recognition',
    position: 'top_left',
    config: {
        // 1=LBPH | 2=Fisher | 3=Eigen
        recognitionAlgorithm: 1,
        // Threshold for the confidence of a recognized face before it's considered a
        // positive match.  Confidence values below this threshold will be considered
        // a positive match because the lower the confidence value, or distance, the
        // more confident the algorithm is that the face was correctly detected.
        lbphThreshold: 50,
        fisherThreshold: 250,
        eigenThreshold: 3000,
        // force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
        useUSBCam: false,
        // Path to your training xml
        trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
        // recognition intervall in seconds (smaller number = faster but CPU intens!)
        interval: 2,
        // Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
        logoutDelay: 15,
        // Array with usernames (copy and paste from training script)
        users: ['zlp'],
        //Module set used for strangers and if no user is detected
        defaultClass: "default",
        //Set of modules which should be shown for every user
        forAllClass: "for_all"
		    }
		},*/</code></pre>

paviro, 2016-11-15:
Really does not make sense :/ Can you try running ONLY this module in a new MM install?

paviro, 2016-12-04:
I will close this for now feel free to reopen.

closed 2016-12-04 by paviro