paviro/MMM-Facial-Recognition #18: OpenCV Version

dr4ke616, 2016-09-11:
Hey,

Im wondering if its worth mentioning what version of OpenCV this project is compatible with. I installed OpenCV `3.1.0`, and had some issues. One of which being `cv2.createLBPHFaceRecognizer` in the facerecognition.py script needed to be called from a `face` module, such as `cv2.face.createLBPHFaceRecognizer`.

Another issue is in `3.x` versions `model.predict` doesnt return a tuple containing the label and confidence score, but rather it just returns the label. 

Even better would it be possible to have it support both `2.x` and `3.x` versions. I am aware there are some backward compatibility issues when using `3.x` from `2.x`, so perhaps just a statement in the README saying which version of OpenCV is intended to be used?

iPython output:

```python
In [1]: import cv2; cv2.__version__
Out[1]: '3.1.0'
```

paviro, 2016-10-03:
I will check :)

paviro, 2016-11-20:
Any chance you could have a look how to add support for both?