paviro/MMM-Facial-Recognition #19: PiCam vs Normal USB Cam

Rian9771, 2016-10-23:
Hi,

I got both of your repo's downloaded and i am int he process of training/capturing images from the webcam. When i do python capture.py, i get the following messages.

python capture.py
What do you want to do?
[1] Capture training images from webcam
[2] Convert '*.jpg' pictures from other cameras to training images
--> 1

Enter the name of the person you want to capture or convert images for.
--> Test
Images will be placed in ./training_data/Test

--------------------
Starting process...

Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:EN                                         OMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/MMM-Facial-Recogniti                                         on-Tools/facetrainer/lib/picam.py", line 24, in run
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 415, in __ini                                         t__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 444, in _init                                         _camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure                                          that the camera has been enabled.

Do i have to use the PiCam or can i use any other USB webcam? I also did enable the cam using raspi-config. 

Any help would be appreciated. 

Thanks

Rian9771, 2016-10-25:
Got it working!

closed 2016-10-25 by Rian9771

andrewgamalkamel, 2019-05-13:
how did you made it with us webcam?