paviro/MMM-Facial-Recognition #20: Face Recognition Not starting?

JoeyRentenaar, 2016-11-02:
Hello, First of all thanks a lot for building this great module!

I am trying to get it to work, I've installed the module and it looks like It doens't start:

Output when I start the NPM:
`pi@raspberrypi:~/MagicMirror $ npm start
> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

Loading config ...
Loading module helpers ...
Initializing new module helper ...
Initializing new module helper ...
All module helpers loaded.
Starting server op port 8080 ... 
Server started ...
Connecting socket for: camera
camera helper started ...
Connecting socket for: MMM-Facial-Recognition
Staring module helper: MMM-Facial-Recognition
Sockets connected & modules started ...
Launching application.
initializing nodemailer
`
It looks like its installed fine, because of the starting module helper: MMM-Facial-Recognition.
But I am not getting the normal output like I see by other people:
[MMM-Facial-Recognition] Facerecognition started...
[MMM-Facial-Recognition] Loading training data...

My Config.json file:
`/* Magic Mirror Config Sample
 *
 * By Tom & Joey
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'nl',
	timeFormat: 24,
	units: 'metric',

	modules: [

		{
	            module: 'camera',
		    position: 'top_center',
		    config: { 
			selfieInterval: 3,  // Time interval in seconds before the photo will be taken.
		        emailConfig: {
		            service: 'Hotmail', // Email provider to use to send email with a photo.
		            auth: {
		                user: '<s1084836@hotmail.com>', // Your email account
		                pass: '<test12345>'        // Your password for email account
		            }
		        }
		}
		},

		{
		    module: 'MMM-Facial-Recognition',
		    position: 'top_left',
		    config: {
			        // 1=LBPH | 2=Fisher | 3=Eigen
				recognitionAlgorithm: 1,
				// Threshold for the confidence of a recognized face before it's considered a
				// positive match.  Confidence values below this threshold will be considered
				// a positive match because the lower the confidence value, or distance, the
				// more confident the algorithm is that the face was correctly detected.
				lbphThreshold: 50,
				fisherThreshold: 250,
				eigenThreshold: 3000,
				// force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
				useUSBCam: true,
				// Path to your training xml
				trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
				// recognition intervall in seconds (smaller number = faster but CPU intens!)
				interval: 2,
				// Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
				logoutDelay: 15,
				// Array with usernames (copy and paste from training script)
				users: ['Tom'],
				//Module set used for strangers and if no user is detected
				defaultClass: "default",
				//Set of modules which should be shown for every user
				everyoneClass: "everyone"
                            }
					},

		]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
`

Is there anything I'm doing wrong ? :D

paviro, 2016-11-10:
Try removing the camera module :) Could be that they interfere.

JoeyRentenaar, 2016-11-10:
Hey Thanks a lot for the response! I really want this to work haha.

I've removed the camera module and stil getting the same thing :(

paviro, 2016-11-15:
Does the tester do anything?

JoeyRentenaar, 2016-11-15:
Yea, the training and test works. The program recognizes my face and displays my name. But I seems like it wont start when I bootup the Magic Mirror. I am kinda clueless about what I can try

paviro, 2016-11-20:
That's really strange. Please try installing anything from scratch again if you can!

paviro, 2016-12-04:
I close this for now feel free to reopen.

closed 2016-12-04 by paviro