paviro/MMM-Facial-Recognition #23: User undefined

toniweser, 2016-11-16:
Hi, 

first of all: thank you for this module!
The facerecognition_tester is working well. It detects my face and is writing down my name. 

I copied the training.xml to the MMM-Facial_Recoginition folder. Then I started the mirror. The module seems to start and it is also working, but it can't detect my face. 

magicmirror@2.0.0 start /home/pi/magicmirror
electron js/electron.js

Loading config ...
Loading module helpers ...
No helper found for module: alert.
No helper found for module: clock.
Initializing new module helper ...
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
No helper found for module: helloworld.
Initializing new module helper ...
Initializing new module helper ...
All module helpers loaded.
Starting server op port 8080 ... 
Server started ...
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-jokes
Starting node helper for: MMM-jokes
Connecting socket for: MMM-Facial-Recognition
Staring module helper: MMM-Facial-Recognition
Sockets connected & modules started ...
Launching application.
Create new calendar fetcher for url: http://xyz... - Interval: 300000
Create new news fetcher for url: http://www.spiegel.de/schlagzeilen/tops/index.rss - Interval: 300000
Create new joke fetcher for url: http://api.icndb.com/jokes/random - Interval: 600000
[MMM-Facial-Recognition] Facerecognition started...
[MMM-Facial-Recognition] Loading training data...
[MMM-Facial-Recognition] ALGORITHM: LBPH
[MMM-Facial-Recognition] Training data loaded!
[MMM-Facial-Recognition] --------------------
[MMM-Facial-Recognition] PiCam ausgewählt...
[MMM-Facial-Recognition] User undefined with confidence null logged in.

All it says is **User undefined with confidence null logged in.**

Do you have any ideas?
Thank you very much!

paviro, 2016-11-16:
Have you tried adjusting the thresholds in the config?

toniweser, 2016-11-23:
Hi!
Great! Now it's working fine!
Thank you very much!

closed 2016-12-04 by paviro