paviro/MMM-Facial-Recognition #24: Classes everyone and default

toniweser, 2016-11-23:
Hi there!

I've got a question concerning the classes for different user in config file. 
I have the following behavior:

- class 'default': module is shown when no user is detected and when a stranger is detected. That means, when a known user is detected, the module hides.
- class 'everyone': module is NEVER shown
- classes: 'everyone default': module is shown all the time. That means, when no user is detected, when a stranger or a known person is detected.

Is that behavior right? 

I am looking for a class which is shown, only when a known person or a stranger is detected. Is that possible?

Thank you very much for your help!

Labzz, 2016-11-28:
Well you're searching for a known person. That must be your own name, if you followed the instructions correctly, you trained your facial recog data. The class can look like ` 'your_name', default`. I Hope to helped you further.

toniweser, 2016-11-30:
Thanks for your answer. 
I know how to use the classes. The problem is, that I want to use one class, which is only shown, when a known person or a stranger is detected.
For example when you use `'myname' 'default'` the module is shown when no user is detected and when I am detected. I'm looking for a class which is only shown when a known person (me) and a stranger person is detected.

Labzz, 2016-11-30:
Isn't, 'everyone' , 'myname' 

BTW is it possible to send your source code? From the MMM module ? Seems like the latest changes in the code has broke some scripts.

Labzz, 2016-11-30:
Isn't it?*
Can't edit in phone lel

paviro, 2016-12-04:
@Labzz What did it break?
@toniweser why do you need to show something different if no one is in front of the mirror than if someone is? No one would see it anyway or am I mistaken?

Labzz, 2016-12-04:
@paviro 
The weird thing is that used capture.py option 2(convert) earlier and it worked well. 

After reinstalling my raspberry pi, and the changes you made(which i found on github) it seems like option 1 is working (but my summy on picam didn't) and USB neither works. option 2 didn't react after I choice a location of the picture. More details I'll send you this evening.

toniweser, 2016-12-05:
@paviro Our usecase is like this: 
- We want to turn off the mirror, when no one is standing in front of him. 
- When a known person is detected, we want to show his/her modules.
- When a stranger is detected, we want to show a default set of modules.

In other words: We want to use the camera also as an "on/off switch".

paviro, 2016-12-05:
But that's not possible by showing another module? There already are modules to do that I think.

toniweser, 2016-12-05:
Ah ok. I didn't know that. I will try to find one. Thank you!

closed 2017-01-12 by paviro