paviro/MMM-Facial-Recognition #25: Capture.py option [2] give no reaction

Labzz, 2016-11-28:
Hi Paviro, 

I'm started clean again to remove al the useless moves i made while testing. 
But when i doing exacly the same as before, running `capture.py`, and select option [2] to convert images. 
He don't do anything, the annyoing part is that he doens't give an error :| 
check out the screenshot. 

Do you have experience with the issue ? and is there any log folder? where i can trackback errors to search specific?

Any suggestion's are usefull! 
Thanks :)

[image]

paviro, 2016-11-30:
What format do the images have?

Labzz, 2016-11-30:
Check the attachment I added. But it is, jpg.

vanimesh76, 2016-12-03:
The training and recognition algorithm works best if they are done using the same camera, preferably pi-Camera.

paviro, 2016-12-04:
@Labzz could you send me the pics so I can do some tests?
@vanimesh76 correct :)

Labzz, 2016-12-04:
Ill send you my selfie Collection later Today, what is your email adress?

paviro, 2016-12-04:
https://paulvincentroll.com/impressum.html

Labzz, 2016-12-04:
@paviro I've found a funny but interesting thing ^_^ and it fixed my issue lol! 

PROTIP
-----------------------------------------

If you're running option [2] from the `capture.py` in the training module. 

Don't use folder path like this --> ~/your_path/
But simple use this --> /home/pi/your_path/

I think the ~ symbol ruined the script. LMAO 

Anyways thanks for you're help pivaro

closed 2016-12-04 by Labzz

paviro, 2016-12-04:
That's interesting haven't thought about that! Could you add a warning to the Readme?