paviro/MMM-Facial-Recognition #26: Cannot find module 'python-shell'

YaarPatandarAA, 2016-11-29:
I have no Idea what is happening.

```
pi@raspberrypi:~/MagicMirror $ DISPLAY=:0 npm start

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

Loading config ...
Loading module helpers ...
WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'python-shell'
Loading module helpers ...
App threw an error during load
Error: Cannot find module 'python-shell'
    at Module._resolveFilename (module.js:455:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron-prebuilt/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:4:21)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
Whoops! There was an uncaught exception...
{ Error: Cannot find module 'python-shell'
    at Module._resolveFilename (module.js:455:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron-prebuilt/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:4:21)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12) code: 'MODULE_NOT_FOUND' }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
```

After this Electron opens up and is just a big white full screen of nothing.

I have everything in the config.js commented out except the Facial Recognize module.

```
  /* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'en',
	timeFormat: 12,
	units: 'imperial',

	modules: [
		
		
		/*{
			module: 'alert',
			//classes: 'default everyone',
		},
		{
			module: 'clock',
			position: 'top_left'
			//classes: 'default everyone',
		},
		{
			module: 'calendar',
			header: 'US Holidays',
			position: 'top_left',
			//classes: 'default everyone',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics'
					}
				]
			}
		},
		//{
			//module: 'compliments',
			//position: 'lower_third'
		//},
		{
			module: 'currentweather',
			position: 'top_right',
			//classes: 'default everyone',
			config: {
				location: 'Sacramento',
				locationID: '5389489',  //ID from http://www.openweathermap.org
				appid: 'eee1044dae24d993e5e7835f652e97dc'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			header: 'Weather Forecast',
			//classes: 'Amarjit',
			config: {
				location: 'Sacramento',
				locationID: '5389489',  //ID from http://www.openweathermap.org
				appid: 'eee1044dae24d993e5e7835f652e97dc'
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_left',
			//classes: 'default everyone',
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
		{
       			module: 'stocks',
        		position: 'bottom_bar',
			//classes: 'Amarjit',
        		config: {
            			stocks: '.DJI,MSFT,AAPL,GOOG,INTC,CICS,TSLA,FB', // stock symbols
            			updateInterval: 37000 // update interval in milliseconds
        		}
    		},
		//{
            		//module: 'MMM-Stock',
            		//position: 'bottom_left',
            		//config: {
                		//companies: ['MSFT', 'GOOG', 'ORCL', 'FB']
            		//}
		//},
		//{
		    	//module: 'MMM-Globe',
		    	//position: 'center',
		    	//config: {
		        	//style: 'geoColor',
		        	//imageSize: 400,
		        	//ownImagePath:'',
		        	//updateInterval: 10*60*1000
		    	//}
		//},*/
		{
    			module: 'MMM-Facial-Recognition',
    			config: {
			        // 1=LBPH | 2=Fisher | 3=Eigen
			        recognitionAlgorithm: 1,
			        // Threshold for the confidence of a recognized face before it's considered a
			        // positive match.  Confidence values below this threshold will be considered
			        // a positive match because the lower the confidence value, or distance, the
			        // more confident the algorithm is that the face was correctly detected.
			        lbphThreshold: 50,
			        fisherThreshold: 250,
			        eigenThreshold: 3000,
			        // force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
			        useUSBCam: true,
			        // Path to your training xml
			        trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
			        // recognition intervall in seconds (smaller number = faster but CPU intens!)
			        interval: 2,
			        // Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
			        logoutDelay: 15,
			        // Array with usernames (copy and paste from training script)
			        users: ['Amarjit'],
			        //Module set used for strangers and if no user is detected
			        defaultClass: "default",
			        //Set of modules which should be shown for every user
			        everyoneClass: "everyone"
    			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

```

paviro, 2016-11-30:
Are you sure you followed the instructions? Do a 'npm install' in the module directory!

YaarPatandarAA, 2016-12-01:
Yeah, that fixed it. I just assumed that it would work like the other modules without having to install anything. I thought python shell was same thing as the python that was already installed.

closed 2016-12-01 by YaarPatandarAA