paviro/MMM-Facial-Recognition #28: Tries to load Webcam instead of PiCam...

Snille, 2016-12-29:
Hi, I have installed a MM2 instance on a Jessie Lite, now trying to get the Face Recognition to work (again). It did work on the Jessie (normal) install. But I want to go "lite". :)
What happens is that it tries to load the "Webcam" camera instead of the "Picam".
`[MMM-Facial-Recognition] Training data loaded!
[MMM-Facial-Recognition] --------------------
[MMM-Facial-Recognition] Webcam ausgewählt...
Whoops! There was an uncaught exception...
{ Error: TypeError: Required argument 'device' (pos 1) not found
    at PythonShell.parseError (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_modules/python-shell/index.js:183:17)
    at terminateIfNeeded (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_modules/python-shell/index.js:98:28)
    at ChildProcess.<anonymous> (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_modules/python-shell/index.js:88:9)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
    ----- Python Traceback -----
    File "modules/MMM-Facial-Recognition/facerecognition/facerecognition.py", line 63, in <module>
      camera = config.get_camera()
    File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/facerecognition/config.py", line 57, in get_camera
      return webcam.OpenCVCapture(device_id=0)
    File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/facerecognition/webcam.py", line 31, in __init__
      self._camera.open()
  traceback: 'Traceback (most recent call last):\n  File "modules/MMM-Facial-Recognition/facerecognition/facerecognition.py", line 63, in <module>\n    camera = config.get_camera()\n  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/facerecognition/config.py", line 57, in get_camera\n    return webcam.OpenCVCapture(device_id=0)\n  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/facerecognition/webcam.py", line 31, in __init__\n    self._camera.open()\nTypeError: Required argument \'device\' (pos 1) not found\n',
  executable: 'python',
  options: null,
  script: 'modules/MMM-Facial-Recognition/facerecognition/facerecognition.py',
  args: [ '{"recognitionAlgorithm":1,"lbphThreshold":50,"fisherThreshold":250,"eigenThreshold":3000,"useUSBCam":false,"trainingFile":"modules/MMM-Facial-Recognition-Tools/facetrainer/training.xml","interval":1,"logoutDelay":30,"users":["Snille","Louise","Martin"],"defaultClass":"default","everyoneClass":"everyone"}' ],
  exitCode: 1 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
`
Same happens with the training module.
The training data that it loads comes from the other installation (copied it to not need to do it all again).

I have even added the "useUSBCam: false" parameter in the config.js to force it not to load the webcam... 

If I use the Picamera with: raspistill -v -o test.jpg
I see the "camera feed" on screen and get the test picture, so the pi camera module is working.

What have I forgotten?

Thank you for the module! :)

Best regards Erik

paviro, 2016-12-30:
Can you check whether the Pi-Cam libraries are available within python?

Snille, 2016-12-30:
Ah, thank you!

sudo apt-get install python-picamera

solved the problem.
It was not installed. :)
Working great now. Case closed.

closed 2016-12-30 by Snille

paviro, 2016-12-30:
Glad to hear! :)