paviro/MMM-Facial-Recognition #35: Everyone class not working - FIXED

Matze213, 2017-01-17:
Hi, our everyoneClass was not working. Being more precise, it didn't load the modules which were tagged for everyone.
To fix it you need to add just two little things in the MMM-Facial-Recognition.js:

Line 90 to 94 old code:
`if (payload.user == -1){
	this.current_user = this.translate("stranger")
        this.current_user_id = payload.user;
}`

Our "fixes":
`if (payload.user == -1){
	this.current_user = this.translate(this.config.everyoneClass)
	this.current_user_id = payload.user;
	this.login_user()
}`

If you want to still have the login message being "Hello Stranger" you need to name your everyoneClass "Stranger"

Greetings
Matze

stanchanpsu, 2017-01-31:
Why isn't this integrated via a PR?

stanchanpsu, 2017-01-31:
I don't think this is the correct fix. The correct fix should be line 76 change from:

`MM.getModules().withClass(this.config.defaultClass).exceptWithClass(this.config.everyoneClass).enumerate(function(module) {`

to

`MM.getModules().withClass(this.config.defaultClass).enumerate(function(module) {`

Because when you logout_user, the default classes should show regardless if it also has the everyone class.

Then lines 90-94 can remain the same.

Matze213, 2017-02-14:
True, was just a small try form myself. But you are right.

paviro, 2017-04-02:
Please define not working :) It does work together with the default class doesn't it? Not sure why I did this but I had a reason back then. If you have a better approach send a PR :)

da4throux, 2017-05-27:
To me the issue is easily visible for a module which only has the 'everyone' class... It does not show when a user is logged in

I created PR #46 to solve this.

paviro, 2018-04-25:
I am sorry for my late response... I will look into your PR.