paviro/MMM-Facial-Recognition #38: Facial-Recognition not recognizing my face

JoeyRentenaar, 2017-01-31:
Hello,

Thanks for building this great module. When I run the "npm start" I keep getting the following message:
```
[MMM-Facial-Recognition] Webcam Found...
[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.
[MMM-Facial-Recognition] User undefined with confidence null logged in.
[MMM-Facial-Recognition] User undefined logged out.
[MMM-Facial-Recognition] User undefined with confidence null logged in.
```
Sometimes it regoonizes my face, but it is like a 1/1000 change. I'm wondering why, when I run the test script my it recognizes my face really fast and always. I uploaded like 350 photo's. I've tryed diffrend back grounds, with enough light.

I just dont understand why it won't see my face while the test script is 95% of the time right. 

My config File:
`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'en',
	timeFormat: 24,
	units: 'metric',

	modules: [

		{
			module: 'clock',
			position: 'top_left',
			classes: 'default',
		},

		
		{
		    module: 'MMM-Facial-Recognition',
		    classes: 'default',
			config: {
		        // 1=LBPH | 2=Fisher | 3=Eigen
		        recognitionAlgorithm: 1,
		        // Threshold for the confidence of a recognized face before it's considered a
		        // positive match.  Confidence values below this threshold will be considered
		        // a positive match because the lower the confidence value, or distance, the
		        // more confident the algorithm is that the face was correctly detected.
		        lbphThreshold: 50,
		        fisherThreshold: 250,
		        eigenThreshold: 3000,
		        // force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
		        useUSBCam: true,
		        // Path to your training xml
		        trainingFile: '/home/pi/Desktop/MagicMirror/modules/MMM-Facial-Recognition/training.xml',
		        // recognition intervall in seconds (smaller number = faster but CPU intens!)
		        interval: 1,
		        // Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
		        logoutDelay: 5,
		        // Array with usernames (copy and paste from training script)
		        users: ['test', 'tom', 'Tom'],
		        //Module set used for strangers and if no user is detected
		        defaultClass: "default",
		        //Set of modules which should be shown for every user
		        everyoneClass: "everyone"
			
		    }
		},
		

		{
		        module: 'helloworld',
		        position: 'lower_third', // This can be any of the regions.
			classes: 'default',
		        config: {
		            // See 'Configuration options' for more information.
		            text: 'OMG, look at you!'
		     }
  		},

		{
			module: 'compliments',
			position: 'upper_third',
			classes: 'Tom',
		},

			{
			module: 'newsfeed',
			position: 'bottom_left',
			classes: 'Tom',
			config: {
				feeds: [
					{
						title: "Nu Algemeen nieuws!",
						url: "http://www.nu.nl/rss/Algemeen"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			
			}
		},
		
		
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
`

Patrick-Remy, 2017-02-21:
The test script uses threshold 80 in config. Nonetheless, it is much slower. No idea why.

slametps, 2017-03-09:
You should check your capture as the inputs. I crop all the captured images with my face only (see Negatives) with almost 50 images per user. Using algorithm 2, this module is almost recognize correct user face more than 90%.

estensen, 2017-03-22:
I also had the same problem, but it works much better after increasing the `lbphThreshold` from 50 to 80. It now recognises me most of the time with a confidence between 40 and 60.

paviro, 2017-04-02:
Yeah the tester uses a lasher threshold. Its slower because it does has to recognize you three times before accepting it as a match in order to prevent logouts if one recognition was faulty.

closed 2017-04-02 by paviro

coltono, 2018-02-20:
Is there a way to not have it recognize 3 times? how can you speed up the recognition process? 5 seconds is just a bit too long