paviro/MMM-Facial-Recognition #45: Face Recognition TypeError: 'int' object is not iterable

b00002561, 2017-04-17:
Hi, I seem to be having an issue when trying to implement the face recognition tester. When I run the tester on the file I am getting an error, I cannot seem to resolve it.

This is what displays in terminal:

```
Loading training data...

ALGORITHM: LBPH

Training data loaded!

Picam selected...

Traceback (most recent call last):

File "facerecognition.py", line 61, in <module>

label, confidence = model.predict(crop)

TypeError: 'int' object is not iterable
```

My picamera stays active but I can see that it has stopped on my face in the background, Do you have any idea how to resolve this issue, as I am at a loss here.

inderpartap, 2018-02-25:
@b00002561 There is a bug in the recent build of OpenCV. the predict function does not return confidence anymore.

As a temporary workaround, change the following lines
`label, confidence = model.predict(crop)` 
to 
`label = model.predict(crop)`
`confidence = 0.7`

Here, I am assigning the confidence, a fixed value. Though, if you want to remove the bug, follow this link - [http://answers.opencv.org/question/82294/cant-get-predict-confidence/](url)

paviro, 2018-04-25:
Is this bug still in OpenCV?