paviro/MMM-Facial-Recognition #51: Recognizes faces but not name

ValentinDenis, 2017-06-15:
Hi,

We did everything according to the readme, we added two faces and obtained a Training.xml. But in the training.xml there is no reference to the names we entered. The recognition tool is working fine, it recognizes our faces with our name and a confidence, but once the training file is loaded with MMM-Facial-Recognition, it recognizes my face but not with my name ("Hello stranger" etc)

Did we forget a step or do anything wrong ?

Thanks !

paviro, 2017-06-15:
Hi there,
Hello stranger means it did not recognize you. The module is a bit stricter as to when it accepts a face. Try playing with the threshold.

ValentinDenis, 2017-06-15:
I just had a positive result with LBPH on a threshold of 70 (with 54 confidence). 
We'll keep trying with different algorithm and thresholds.
Could it be that I should train it with more pictures ? My friend and I both have glasses, maybe without them it could lead to better results ?

Thanks for the really quick answer anyway !

paviro, 2017-06-19:
Glasses sadly don't work that well :/

closed 2017-06-19 by paviro