paviro/MMM-Facial-Recognition #52: Recognizes name but doesn't load modules

iAmoric, 2017-08-18:
The facial recognition works well, I can see in the console some messages like

> User <name> with confidence xxxx logged in

But modules aren't loaded, nothing changes on the mirror.

My config is the following : 

```javascript
{
	module: 'MMM-Facial-Recognition',
	config: {
                recognitionAlgorithm: 1,
		lbphThreshold: 50,
		fisherThreshold: 250,
		eigenThreshold: 3000,
	        useUSBCam: false,
		trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
		interval: 2,
		logoutDelay: 15,
		users: ['<name>'],
		defaultClass: 'default',
		everyoneClass: 'everyone',
		welcomeMessage: true
	}
},
```
and for example : 

```javascript
{
	module: 'clock',
        position: top-center,
        classes: 'default everyone'
},
{
	module: 'compliments',
        position: lower_third,
        classes: '<name>'
},
```

The module 'clock' works correctly, but the module 'compliments' doesn't appear when my faces is detected and recognized.

What am I doing wrong ?

inderpartap, 2018-02-25:
`users: ['<name>'], and classes: '<name>'
`
Here, write your own name replacing <name>. It should match with the one you trained it with.

paviro, 2018-04-25:
I will close this issue for now. If you still have problems feel free to reopen / ask on the forum.

closed 2018-04-25 by paviro