paviro/MMM-Facial-Recognition #55: questions about adding to the config.js

diverdennis, 2018-04-26:
Hello,

i added this to my config.js and itßs not working...MagicMirror Start fails:

> 			{
						symbol: "calendar-check-o ",
						url: "webcal://www.ifeiertage.de/rp-.ics"
					}
				]
			}
		},
		{
			module: "MMM-Facial-Recognition",
			config: {
				// 1=LBPH | 2=Fisher | 3=Eigen
				recognitionAlgorithm: 1,
				// Threshold for the confidence of a recognized face before it's considered a
				// positive match.  Confidence values below this threshold will be considered
				// a positive match because the lower the confidence value, or distance, the
				// more confident the algorithm is that the face was correctly detected.
				lbphThreshold: 50,
				fisherThreshold: 250,
				eigenThreshold: 3000,
				// force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
				useUSBCam: true,
				// Path to your training xml
				trainingFile: "/home/pi/MagicMirror/modules/MMM-Facial-Recognition/training.xml",
				// recognition intervall in seconds (smaller number = faster but CPU intens!)
				interval: 2,
				// Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
				logoutDelay: 15,
				// Array with usernames (copy and paste from training script)
				users: [Dennis],
				//Module set used for strangers and if no user is detected
				defaultClass: "default",
				//Set of modules which should be shown for every user
				everyoneClass: "everyone",
				// Boolean to toggle welcomeMessage
				welcomeMessage: true
			}
		},
		{
			module: "compliments",
			position: "lower_third"

> 

Can´t find out whats wrong... Anyboy has an idea about this?

I would appreciate any help.

paviro, 2018-04-26:
What exactly isn’t working? Can’t help you without details. Error log?

diverdennis, 2018-04-26:
thanks for the response.

yes i did add a comma and i can´t find errors with npm run config:check.

just don´t get it..