paviro/MMM-Facial-Recognition #58: TypeError: PythonShell is not a constructor     at Class.python_start (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:11:21)

akashpbhardwaj, 2018-09-13:
The training is working fine but the Module doesnot works when used in config.js 
Please check the issue !

Here is the Log :
npm start

> magicmirror@2.4.1 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.4.1
Loading config ...
Loading module helpers ...
Initializing new module helper ...
Module helper loaded: MMM-Facial-Recognition
Initializing new module helper ...
Module helper loaded: MMM-OnScreenMenu
No helper found for module: MMM-SimpleLogo.
Initializing new module helper ...
Check MagicMirror version for node helper 'MMM-CoinMarketCap' - Minimum version:  2.2.1 - Current version: 2.4.1
Version is ok!
Module helper loaded: MMM-CoinMarketCap
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
No helper found for module: MMM-google-route.
Initializing new module helper ...
Module helper loaded: newsfeed
No helper found for module: MMM-connection-status.
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: MMM-Facial-Recognition
Starting module helper: MMM-Facial-Recognition
Connecting socket for: MMM-OnScreenMenu
Connecting socket for: MMM-CoinMarketCap
MMM-CoinMarketCap: module loaded! Path: /home/pi/MagicMirror/modules/MMM-CoinMarketCap
MMM-CoinMarketCap: Path "/logos" configured successfully.
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Launching application.
Whoops! There was an uncaught exception...
TypeError: PythonShell is not a constructor
    at Class.python_start (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:11:21)
    at Class.socketNotificationReceived (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:40:14)
    at Socket.<anonymous> (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:113:11)
    at emitTwo (events.js:126:13)
    at Socket.emit (events.js:214:7)
    at /home/pi/MagicMirror/node_modules/socket.io/lib/socket.js:528:12
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
MMM-CoinMarketCap: Socket Notification Received: "INIT".
MMM-CoinMarketCap: Socket Notification Received: "GET_LISTINGS".
Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/33/India_Holidays.ics - Interval: 300000
Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
MMM-CoinMarketCap: Socket Notification Received: "GET_CURRENCY_DETAILS".
MMM-CoinMarketCap: Socket Notification Received: "GET_CURRENCY_DETAILS".
MMM-CoinMarketCap: Socket Notification Received: "GET_CURRENCY_DETAILS".
MMM-CoinMarketCap: Socket Notification Received: "GET_CURRENCY_DETAILS".

2nd The config file
var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	language: "en",
	timeFormat: 24,
	units: "metric",

	modules: [
{
	module: 'MMM-Facial-Recognition',
	config: {
		// 1=LBPH | 2=Fisher | 3=Eigen
		recognitionAlgorithm: 1,		
		lbphThreshold: 50,
		fisherThreshold: 250,
		eigenThreshold: 3000,
		// force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
		useUSBCam: false,
		// Path to your training xml
		trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
		// recognition intervall in seconds (smaller number = faster but CPU intens!)
		interval: 2,
		// Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
		logoutDelay: 15,
		// Array with usernames (copy and paste from training script)
		users: ['Akash', 'None'],
		//Module set used for strangers and if no user is detected
		defaultClass: "default",
		//Set of modules which should be shown for every user
		everyoneClass: "everyone",
		// Boolean to toggle welcomeMessage
		welcomeMessage: true
	}
},
		{
	            module: 'MMM-OnScreenMenu',
	            position: 'bottom_right',
	            /* Valid positions: 'top_right', 'top_left', 'bottom_right', 'bottom_left' */
	            config: {
	                touchMode: true,
	                enableKeyboard: true,
	                // ... see more options below
	      		    }
     		 }
]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

Other modules are also there but i think no need to show here as the error in the log is from Face recognition module

PimDietzenbacher, 2018-09-28:
Same problem. How did you solve this????

dimitrystd, 2018-10-03:
Me too. JIC
```
MagicMirror: v2.4.1
Python: 2.7.13
Node: 9.11.2
NPM: 6.4.1
```

JooMiHyun, 2018-10-03:
i also have same problem. How did you solve this?

ejwessel, 2018-10-06:
+1 having this issue

shayfer, 2018-10-26:
I have tried that as well but now I get a new error.

SyntaxError: Unexpected token T in JSON at position 0
    at JSON.parse (<anonymous>)
    at PythonShell.asJson (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:350:21)
    at /home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:302:42
    at Array.forEach (<anonymous>)
    at PythonShell.recieveInternal (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:298:15)
    at PythonShell.receiveStderr (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:282:21)
    at Socket.<anonymous> (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:100:18)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
{ Error: ImportError: No module named 'cv2'
    at PythonShell.parseError (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:232:21)
    at terminateIfNeeded (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:121:32)
    at ChildProcess.<anonymous> (/home/pi/MagicMirror/modules/MMM-Motion-Detection/node_modules/python-shell/index.js:113:13)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
    ----- Python Traceback -----
    File "modules/MMM-Motion-Detection/motiondetection/motiondetection.py", line 19, in <module>
      import cv2
  traceback: 'Traceback (most recent call last):\n  File "modules/MMM-Motion-Detection/motiondetection/motiondetection.py", line 19, in <module>\n    import cv2\nImportError: No module named \'cv2\'\n',
  executable: 'python3',
  options: null,
  script: 'modules/MMM-Motion-Detection/motiondetection/motiondetection.py',
  args: 
   [ '{"useUSBCam":false,"interval":1,"motionStopDelay":120,"detectionThreshold":1000,"turnOffDisplay":true}' ],
  exitCode: 1 }

nater303, 2018-11-03:
- Open node_helper.js
var option = {pythonPath: 'python'}
    const pyshell = new PythonShell('modules/' + this.name + '/facerecognition/facerecognition.py', option, { mode: 'json', args: [JSON.stringify(this.config)]});

nosee818, 2018-11-13:
> * Open node_helper.js
>   var option = {pythonPath: 'python'}
>   const pyshell = new PythonShell('modules/' + this.name + '/facerecognition/facerecognition.py', option, { mode: 'json', args: [JSON.stringify(this.config)]});

Do you mean add var options? It does not work for me. :(

roudaille, 2018-11-21:
same issue over here.  Same node_helper.js tested with MMM-Facial-Recognition-OCV3 is working fine.

fpdrozd, 2018-12-08:
Hi, I installed an older version of python-shell ([0.4.0 - released 3 years ago](https://www.npmjs.com/package/python-shell/v/0.4.0)) and it works without modifying the node_helper.js file. There had to be some breaking changes.

Sumrag, 2019-02-21:
> Hi, I installed an older version of python-shell ([0.4.0 - released 3 years ago](https://www.npmjs.com/package/python-shell/v/0.4.0)) and it works without modifying the node_helper.js file. There had to be some breaking changes.

Hi! How did you install an older version of python shell to make the MMM-Facial-Recognition work?

Sumrag, 2019-02-22:
> I have tried that as well but now I get a new error.
Where you able to solve the issue? My output is the same as yours when I insert the curly brackets.

fpdrozd, 2019-02-27:
@Sumrag You can install this specific version of python-shell by this command:
`npm install python-shell@0.4.0`

([npm docs reference](https://docs.npmjs.com/cli/install))