paviro/MMM-Facial-Recognition #59: Not sure if mine installed or not

Grahamrich1, 2018-10-14:
I am using a MagicMirror install that Ive been working on for a while now.  I've run "npm install python-shell" and "sudo apt-get install libopencv-dev python-opencv" which seem to have run ok.  I've installed my camera (Onboard Pi camera) and tested it.  Seems to take pictures of this ugly guy.
I've updated my Config.js file and to test, I added a line "classes: 'Graham-Rich' " to one of the module I only want to appear when I'm in front of the mirror.  Because I've not getup a class 'Graham-Rich' yet I wasn't expecting to see this module, but I can.  Is it because its defaulting to display because the class doesn't exist or do I have bigger issues

Grahamrich1, 2018-10-14:
Just noticed, the Config.js file refences a "trainingFile: 'modules/MMM-Facial-Recognition/training.xml' " but I don't have this directory.  Have I missed some of the install off

Grahamrich1, 2018-10-14:
So now I am getting a black screen, not the one that is saying you have an issue with your config file.  When I remove the following is works again

	module: 'MMM-Facial-Recognition',
	config: {
		// 1=LBPH | 2=Fisher | 3=Eigen
		recognitionAlgorithm: 1,
		// Threshold for the confidence of a recognized face before it's considered a
		// positive match.  Confidence values below this threshold will be considered
		// a positive match because the lower the confidence value, or distance, the
		// more confident the algorithm is that the face was correctly detected.
		lbphThreshold: 50,
		fisherThreshold: 250,
		eigenThreshold: 3000,
		// force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
		useUSBCam: false,
		// Path to your training xml
		trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
		// recognition intervall in seconds (smaller number = faster but CPU intens!)
		interval: 2,
		// Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
		logoutDelay: 15,
		// Array with usernames (copy and paste from training script)
		users: [],
		//Module set used for strangers and if no user is detected
		defaultClass: "default",
		//Set of modules which should be shown for every user
		everyoneClass: "everyone",
		// Boolean to toggle welcomeMessage
		welcomeMessage: true
	}

jpmorpixels, 2019-01-14:
Did you fix it?

slimcdk, 2019-02-13:
You need to train the model on your face and then link to the trained weights.
https://github.com/paviro/MMM-Facial-Recognition-Tools